Denvi/Candle #649: Cnc works randomly when send g-code

Wita-blip, 2024-10-23:
**Versions**
 - Candle version: 1.0.11
 - Firmware version: 0.9j

**PC info**
- OS:windows 10 64bit
- OpenGL version: idk

**Describe the bug**
candle 1.01.1 looks good, manual jogging works fine, homing is good, manual jogging works fine, after i set X0, Y 0 and Z0 and i click send, cnc runs randomly. os window 10 64bit, grbl 0.9j, candle 1.0.1.1, need help

Wita-blip, 2024-10-23:
when using candle 1.0.10 vizualizer getting blank but manual jogging, homing works fine

Wita-blip, 2024-10-23:
@Denvi  need help

mar0x, 2024-10-23:
Hello,
You are using outdated GRBL firmware version and this forces you to stay on outdated Candle version.
Try latest Candle 1.1.7+ without CNC connect (visualizer). If it works, I would recommend to upgrade your CNC firmware up to 1.1h and use latest Candle 1.1.7+.

Wita-blip, 2024-10-23:
ok, i will try it, thanks for the info

Wita-blip, 2024-10-23:
I will let you know the result, thank you very much for the information

Wita-blip, 2024-10-23:
[image]

Wita-blip, 2024-10-23:
one more question, I bought a cnc with a control board from the manufacturer as pictured (firmware grbl 0.9j) can the control board as pictured be updated to grbl 1.1 firmware? if possible, what should I choose? (Is it arduino uno or arduino nano)

Wita-blip, 2024-10-23:
[image]

mar0x, 2024-10-23:
As far as I can see, this is not an Arduino board (Uno nor Nano), but CNC controller.

Could you please share exact board (CNC) name and version, maybe URL with the information (online store, vendor site etc).

If you find upgrade instructions on the vendor site or get it directly from the support, this would be perfect.

Complete detailed board photo may also help (exact symbols from the square 32pin chip required).

Wita-blip, 2024-10-23:
Ok, wait a minute

Hostaler, 2024-10-23:
> Hola, está utilizando una versión de firmware GRBL obsoleta y esto lo obliga a permanecer en la versión obsoleta de Candle. Pruebe la última versión de Candle 1.1.7+ sin conexión CNC (visualizador). Si funciona, recomendaría actualizar el firmware de su CNC hasta 1.1h y utilizar la última versión de Candle 1.1.7+.

Es mejor la version 1.1.7 que la que yo tengo? Yo tengo la 1.2.15b

[image]

Wita-blip, 2024-10-23:
[image]

Wita-blip, 2024-10-23:
[image]

Wita-blip, 2024-10-23:
this is the clearest photo i could get

Wita-blip, 2024-10-23:
Could you please share exact board (CNC) name and version, maybe URL with the information (online store, vendor site etc).

If you find upgrade instructions on the vendor site or get it directly from the support, this would be perfect.

Answer: 
i don't have the info, but i like this control board because it can directly power a 48v spindle with the addition of a dc to dc step up board and easy to handle, just follow the information on the control board for installing limit switches, DC cooling fans, installing 48v spindles with only one power supply`

Wita-blip, 2024-10-23:
Cnc 3018

Wita-blip, 2024-10-23:
written on ic atmel mega 328p u-th 354730 1924he or 1924me

Wita-blip, 2024-10-23:
I just tested my cnc with candle 1.0.12, now I know what causes the random aefara movement, namely the z probe, after X0y0 I do a probe, after the probe I give the command G0x20y20, it moves according to the command, but once I give the command G0X0Y0 it doesn't move at all with the coordinate command given below the number 20 (X20 Y20), after I exit the candle and re-enter, I manually determine the Z0 point, it works normally even with the g code file that will be worked on, it works normally, what should I do so that I can use the z probe to determine the Z0 point not manually

Wita-blip, 2024-10-23:
I just tested my cnc with candle 1.0.12, now I know what causes it to move randomly, namely the z probe, after X0y0 I do a probe, after the probe I give the command G0x20y20, it moves according to the command, but once I give the command G0X0Y0 it doesn't move at all with the coordinate command given below the number 20 (X20 Y20), after I exit the candle and re-enter, I manually determine the Z0 point, it works normally even with the g code file that will be worked on, it works normally, what should I do so that I can use the z probe to determine the Z0 point not manually

mar0x, 2024-10-23:
> Is version 1.1.7 better than the one I have? I have 1.2.15b

@Hostaler best version is the version which works for you and covers your requirements :)

1.1.7 is definitely better than 1.0.11 which seems not perfect for author of this issue. 1.2 is experimental version this why I'm not recommend it.

mar0x, 2024-10-23:
@Wita-blip only first photo uploaded successfully.
"CNC 3018" only describe the size of the working area 30 cm x 18 cm
"atmega 328p" is important information;
Your board upgrade capabilities depends on current board firmware. Firmware with bootloader is easier to upgrade.

Wita-blip, 2024-10-23:
1.1.7 is not imperfect for me, I just compensate for what I have for now that I can fit the control board which is grbl 0.9j so I can only use candle 1.0x and I also don't know how to update the firmware, I'm still learning it

Wita-blip, 2024-10-23:
[image]

Wita-blip, 2024-10-23:
[image]

Wita-blip, 2024-10-23:
[image]

Wita-blip, 2024-10-23:
[image]

Wita-blip, 2024-10-23:
just bought an arduino uno with cnc shield for firmware upgrade

closed 2024-10-24 by Wita-blip