Denvi/Candle #650: bug version 1.2b

ah1102, 2024-10-28:
**Versions**
 - Candle version: 1.2.*
 - Firmware version: 1.1f

**PC info**
- OS: win10

**Describe the bug**

1)There are no tips
[image]

2) Keys do not work in the process - pause and interrupt. Two right buttons in the right in the photo.
[image]

Hostaler, 2024-10-29:
Hello good morning, the same thing happens to me and in the end Denvi recommended me to install version 1.17 , although I see 1.2b as much more complete but for the moment I don't see another solution.

Hostaler, 2024-10-29:
You have a fork of Denvi\Candle here, but I don't know if it works well. 

https://github.com/mar0x/Candle

mar0x, 2024-10-29:
> You have a fork of Denvi\Candle here, but I don't know if it works well.
> 
> https://github.com/mar0x/Candle
> 
> It is also experimental .

I think experimental version is overcomplicated and I almost don't touch it.

Hostaler, 2024-10-30:
> > Tienes una bifurcación de Denvi\Candle aquí, pero no sé si funciona bien.
> > https://github.com/mar0x/Candle
> > También es experimental.
> 
> Creo que la versión experimental es demasiado complicada y casi no la toco.

😞😞

Samer-Amin, 2024-12-14:
> **Versions**
> 
> * Candle version: 1.2.*
> * Firmware version: 1.1f
> 
> **PC info**
> 
> * OS: win10
> 
> **Describe the bug**
> 
> 1)There are no tips ![Снимок экрана 2024-10-28 в 19 51 20](https://private-user-images.githubusercontent.com/58341369/380811898-77905864-dcdd-43bd-9a8b-6ed9a6b41ed8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyMDUxMjcsIm5iZiI6MTczNDIwNDgyNywicGF0aCI6Ii81ODM0MTM2OS8zODA4MTE4OTgtNzc5MDU4NjQtZGNkZC00M2JkLTlhOGItNmVkOWE2YjQxZWQ4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE0VDE5MzM0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJkYjhiNDMxYWMyNmZlYWE0OGY0MmE1MjEzNWU1ZDUyNDM1NjBlM2UzZGQzYTBmN2Q0MTkwZWNiZWU2ZTBhOGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ZknhTCPs2w_VrKqYXpE3T6ar-0PkkOz9qrc1fXkbpow)
> 
> 2. Keys do not work in the process - pause and interrupt. Two right buttons in the right in the photo.
>    ![Снимок экрана 2024-10-28 в 19 52 41](https://private-user-images.githubusercontent.com/58341369/380813038-a77f4730-1cbe-42dc-8116-580f0f984013.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyMDUxMjcsIm5iZiI6MTczNDIwNDgyNywicGF0aCI6Ii81ODM0MTM2OS8zODA4MTMwMzgtYTc3ZjQ3MzAtMWNiZS00MmRjLTgxMTYtNTgwZjBmOTg0MDEzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE0VDE5MzM0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU2NTg3YTkzM2YyMGRlZDhlNTY2MDRiNGUzMDI5Y2YwZTc0NTg0ZDcwZTNiYmVkYjM5ODg0Y2NkMjk5OWVkZmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.VyCqkwmOgjNkJ2FjRaCP5UZUHcLyY304xyGPCt_hg1Y)

These keys are user defined in settings.

paulfjujo, 2025-01-15:
hello,

my config

O.S.:   WIN 10
using YAT PC  terminal 115200bds

Makerbase GRBL ESP32 board V2.0
$I
[VER:1.1.2022051001:]
[OPT:PHSW]
[MSG:Using machine:MKS DLC32]
[MSG:Mode=AP:SSDI=Ma_CNC:IP=192.168.4.1:MAC=AC-15-18-CB-54-85]
ok
ok

i can get Pins status infos with this command
concern
[MSG:X  Axis limit switch on pin GPIO(36)]
[MSG:Y  Axis limit switch on pin GPIO(35)]
[MSG:Z  Axis limit switch on pin GPIO(34)]
[MSG:Probe on pin GPIO(22)]

?
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:3.000,3.000,-3.000>
ok
ok

but with Candle  version 1.2.15b   
$HELP < [HLP:$$ $+ $# $S $L $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H $F $E=err ~ ! ? ctrl-x]
same command **?**
? < ok
Problemo : no information returned !

JoanTheSpark, 2025-05-16:
@paulfjujo 
> but with Candle version 1.2.15b H E L P < [ H L P : $ + # $S $L $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H $F $E=err ~ ! ? ctrl-x] same command **?** ? < ok Problemo : no information returned !

Open a separate issue - if you post additional problems in a different issue tracker it is being ignored.

closed 2025-09-05 by github-actions[bot]