Denvi/Candle #654: Jog buttons not working

Go-Chiefs, 2025-01-25:
I have a problem with a Lunyee 3018 Pro Max I just bought.  I am running Candle on a Windows 10 PC.

    1. When I hit the jog buttons non of the axis move.  The work and machine coordinates change and the status changes from Idle to Jog.
    2. When I hit the spindle button nothing happens.  
    3. Opening and sending an NC file also does not work.  The screen shows the carving being done but nothing happens on the machine.
    4. I have an older machine and I am getting the same results except I am able to turn on the spindle of that machine.
    5. I am getting the same results when using OpenBuilds and Easel

What should I check to try and determine what is causing the problem?  Thanks.

Hostaler, 2025-01-26:
I recommend you uninstall Candle and reinstall the program.

Hostaler, 2025-01-27:
I think you need to install a driver called CH340

https://www.arduined.eu/tag/ch340/

Go-Chiefs, 2025-01-27:
I installed the driver that came with the machine, CH341SER.

Also, I have tried to jog the axis by directly inputing the appropriate Gcode but no luck.

Hostaler, 2025-01-28:
Please install Lasergrbl, and tell me if it works for you.(I find it very strange that after everything you've done Candle doesn't work.)

ondras12345, 2025-01-28:
If the coordinates change, but the machine doesn't move, the issue is almost certainly not in Candle. Candle does not deduce work/machine coordinates itself, it receives them via UART from Grbl. This means everything on the PC side, including the CH340 driver, is working.
If Grbl thinks it's moving (coordinates are changing) but the machine isn't, that sounds like a hardware problem. Grbl doesn't do any form of closed-loop control, it just sends step pulses to the stepper motor drivers and assumes the motor moves accordingly. This sounds like either the motors aren't connected properly, or the power supply is dead, or the stepper drivers are dead.

Hostaler, 2025-01-28:
He may also need to downgrade to version 1.1 as the same thing is happening on the old machine. But I would say it's the power supply as you said.

ondras12345, 2025-01-28:
If it was a version mismatch between Candle and Grbl, jogging would not work, but the coordinates wouldn't change.

Hostaler, 2025-01-30:
Have you solved it yet?

Go-Chiefs, 2025-02-03:
> Please install Lasergrbl, and tell me if it works for you.(I find it very strange that after everything you've done Candle doesn't work.)

I'm still working on it.  I had to step away but I need to resolve this soon or return the machine.
Will Lasergrbl work even though I don't have a laser?
Also, I'm having a hard time understanding why neither machine works.  If the problem is hardware I can't see them both not working,
Thanks to everyone for all your help.

Hostaler, 2025-02-04:
> Si las coordenadas cambian, pero la máquina no se mueve, es casi seguro que el problema no esté en Candle. Candle no deduce las coordenadas de trabajo/máquina por sí mismo, las recibe a través de UART desde Grbl. Esto significa que todo en el lado de la PC, incluido el controlador CH340, está funcionando. Si Grbl cree que se está moviendo (las coordenadas están cambiando) pero la máquina no, eso suena como un problema de hardware. Grbl no realiza ningún tipo de control de bucle cerrado, solo envía pulsos de paso a los controladores del motor paso a paso y asume que el motor se mueve en consecuencia. Esto suena como si los motores no estuvieran conectados correctamente, o la fuente de alimentación estuviera muerta, o los controladores del motor paso a paso estuvieran muertos.

Here another user gives you several options and the option that I see with the most possibilities is that your power supply is dead.

David38350, 2025-02-11:
I have the same problem before I upgrade my CNC Firmware with GRBL 1.1

https://github.com/gnea/grbl/releases/download/v1.1h.20190825/grbl_v1.1h.20190825.hex

Now all working for me, with candle latest version.

Go-Chiefs, 2025-02-12:
Thanks David.  I was not ablel to find the firmware version in Candle but I checked my firmware using OpenBuilds and it said it was v1.1h.20190825.  But since the machine is 3 years old and I have never manually updated the firmware I'm not sure that is correct.
The instructions I found online did not help me find the software version in Candle.  Do you have a link that would help me?

closed 2025-09-05 by github-actions[bot]