Denvi/Candle #657: While bug in src/candle/widgets/dropwidget.cpp

GeorgKZ, 2025-05-30:
**Versions**
 - Candle version: [1.2.15b]

Файл: src/candle/widgets/dropwidget.cpp
Строка: while (y >= yl.at(i) && i < yl.count()) i++;

Из-за неправильной последовательности проверки условий завершения цикла происходит аварийное завершение программы при перетаскивании виджета.

Для повышения удобочитаемости можно оформить в виде цикла for:

            int i;
            for (i = 0; i < yl.count(); ++i) {
                if (y < yl.at(i)) break;
            }

Hostaler, 2025-06-13:
can you write in english?

GeorgKZ, 2025-06-13:
File: src/candle/widgets/dropwidget.cpp
Line: while (y >= yl.at (i) && i < yl.count()) i++;

Due to the incorrect sequence of cycle completing conditions checking, the program crashes during widget dragging.

To increase readability, you can arrange it in the  "for" loop form:

int i;
for (i = 0; i < yl.count(); ++i) {
       if (y < yl.at(i)) break;
}

And another issue here:
File: frmmain.cpp
Function: bool frmMain::eventFilter(QObject *obj, QEvent *event)

Please check ui != nullptr here before using ui->:

        if ((m_senderState != SenderTransferring) && (m_senderState != SenderStopping) 
            && ui->chkKeyboardControl->isChecked() && !ev->isAutoRepeat()) 

Without that check the program sometimes crashes when exiting.

Denvi, 2025-08-22:
Так и не смог воспроизвести баг с while.
В любом случае, правки внес и по while и по eventFilter.

closed 2025-09-16 by Denvi