Denvi/Candle #661: The problem with manual feeding

wolmix123, 2025-07-06:
**Versions**
 - Candle version: 2.1

When manually feeding any of the axes, the motor moves in micro jerks, but everything works fine in the automatic feed. Just yesterday, everything was working fine, but today this problem has appeared.

It says this in the console "$J=G21G91X-0.3333Y0Z0F2000 < ok" (this is written with manual feeding (on the arrows) at a constant speed of 2000 millimeters per minute.

Hostaler, 2025-07-07:
Pero la versión 2.1 no existe.

wolmix123, 2025-07-07:
This
[image]

Hostaler, 2025-07-07:
This link takes you directly to the use of  $J 

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging#how-to-use

Hostaler, 2025-07-07:
But enlarge the window by scrolling with the mouse over the area I have marked in red and there you should have a box to enter the amount you want the axis to move. That way you don't need the $J anymore 

[image]

Hostaler, 2025-07-09:
See if this article solves it for you. When you have the result, share it here.

https://github.com/Denvi/Candle/issues/209

closed 2025-09-05 by github-actions[bot]