Denvi/Candle #666: Candle 1.1 chokes after a while.

DEF7, 2025-09-10:
**Versions**
 - Candle v0.7 and Candle v1.1 (both self-compiled versions because metric isn't an option for me)
 - GRBL 0.9i and GRBL 1.1f

**PC info**
- OS: Win10 64-bit
- OpenGL version: 4.6

**Describe the bug**
Consistently across different machines and computers, grblControl and Candle both start choking after a "large" program has run for a while. I thought this was an issue with the feedrate being too high for how many feed coordinates there were because I replaced my entire machine and the computer that I was running my machine with and am now using Candle v1.1 instead of grblControl 0.7 and it exhibits the same issue. A large program will start OK but after several hours they both start choking. The FPS drops down to just a few and the machine feeds through a few coordinates, then stalls, feeds through a few coordinates, and stalls, and this just gets worse and worse with the pauses becoming longer and longer over the hours. I am convinced now that it's something to do with how grblControl and Candle are both processing G-code commands in the background that is non-optimal.

**To Reproduce**
Load a G-code file that's several megabytes in size, and run it.

**Expected behavior**
I expect it to run as smooth at the end as it does at the beginning.

**Screenshots**
N/A

**Example files**
https://www.dropbox.com/scl/fi/2ovuynn669pmd22rynsou/Colliding20-01-0.75mm-taperball.nc?rlkey=jmlmbldz251nh2ap9tw8px6s0&dl=1

**Additional context**
Again, these are self-compiled builds, because I had to go in and fix all of the code assuming metric due to all my physical tools being in inches. This should not be relevant though. The v1.0 release says that "programs with 5 million lines can be loaded now", but I'm not finding that Candle can even handle actually running a file with only 250k lines.

Cheers!

Denvi, 2025-10-08:
In the latest versions of the program, a mechanism limiting the number of table updates per unit of time has been introduced. That should help.

DEF7, 2025-10-08:
Thanks. I had started setting up the environment for building the newer version(s) of Candle but when I realized I had to also download/install MSVC I opted to just go back to the Qt-only versions instead because that was way less of a headache.

closed 2025-11-15 by github-actions[bot]